googlecreativelab/paper-signals #4: Can't get Google Assistant/Paper Signals Link Completed

tomvdb, 2017-12-03:
Got my countdown paper signal built, setting up the code. Next step is getting the unique 3-word identifier. Google assistant takes me to paper signal that tells me about the paper signal website and then asks me do I want to build a countdown,pants, etc. When I reply, it just sits there until google assistant comes back and says that paper signals isn't available right now, please try again later :(

dfashbaugh, 2017-12-03:
Is this happening to you every time you try? Or does the issue seem intermittent?

dfashbaugh, 2017-12-03:
Sorry, didn't mean to close the issue!
Anyway, I have not been able to recreate the problem. So any more information would be very helpful!

tomvdb, 2017-12-03:
I did that for a long time, and then it just started working. The setup was a bit nightmarish and found a few bugs, but will start another issue for those.

Thanks,

closed 2017-12-03 by tomvdb