googlecreativelab/paper-signals #5: Some issues, notes :)

tomvdb, 2017-12-03:
OK, so i got it working, is pretty cool. Have some cool ideas for new signals. These were the issues/bugs I found:

(1) From the documentation and code it seems that the signal unique id has to be uppercase, turns out it has to be lower case. This is confusing especially with point 2
(2) The unique code it reads back, is said too quickly, especially if I'm trying to write it down. Some of the words also didn't sound right. I ended up checking my google activity logs for the actual words.
(3) When you create a new signal, it asks you to say 'I'm ready' to start back reading the code. I have a bit of a speech impediment (can't say r properly) and it was a nightmare to get past that point. Maybe a simple yes or no to continue for getting code. also maybe repeat the code?
(4) It seemed to get confused sometime, not sure if you guys have access to the discussions, but if you do you will see it.

I get that its a concept/prototype/new thing so issues to be expected, just giving feedback on what I found :)

Countdown Signal is now up and running and will be trying the others in the next couple of days.

T

nicolehe, 2017-12-03:
Thanks for these! One question - did you do this on a Google Home, or the Google Assistant on a phone?

tomvdb, 2017-12-03:
I am doing this directly on google home. here is a video of it in action: https://twitter.com/geekc0der/status/937387314560602112

KacobJeith, 2017-12-04:
Thanks for the feedback @tomvdb! Using the Google Assistant app may help during some of the setup steps, since you can read the signal identifier on the screen. Google Home speaks notoriously quickly, so those identifier words really fly by. We've also added lot's of suggestion chips within the Assistant app to the conversation flow that may help you pass any unintentionally challenging steps, like saying "I'm ready."

Looks like some of the extended voice command set didn't make it into the website instructions, but at any time you can say **"What is the identifier for [Signal Name]?"** and you can hear the signal ID again. 

Would be good to un-capitalize the identifier in the code that make that part less confusing off the bat

nicolehe, 2017-12-04:
@KacobJeith I wonder if the user is setting up for the first time on a Google Home, we should maybe suggest that doing it on the Assistant is easier...

tomvdb, 2017-12-04:
I wanted to originally use assistant, but couldnt get it working on my phone, but good to know. Found another problem/bug. When you config the countdown timer with a "Start countdown timer for 30 seconds', then it works fine, but you cant restart a countdown timer for 30 seconds. ie, if I ask for count down to 30 seconds again then it ignores it. I need to start a countdown for another value, then it works again, but not a second time with the same value. hope that makes sense?

PR #18 (2017-12-15): Timer Will Countdown Even When Given Same Duration
This fixes #5 

The countdown signal will now count anytime it is asked to set a time and no longer depends on the time changing.
closed 2017-12-15 by nicolehe