googlecreativelab/paper-signals #8: connecting to gweb-paper-signals.firebaseio.com failed

mintisan, 2017-12-04:
I use my iPhone as a hotspot, and I can connect to `console.firebase.google.com` with my iPhone.

ESP8266 UART message showed that WiFi connection is fine.
```
............
WiFi connected
IP address: 172.20.10.2
connecting to gweb-pape
```

but  I can't connect to firebase  as below:
```
connecting to gweb-paper-signals.firebaseio.com
connection failed
parseObject() failed
Default
```

[image]

Did I miss any settings? 

My modification of code is just as below:

[image]

tomvdb, 2017-12-04:
Not sure if it's the same, but I had the same problem. The esp8266 calls this url:

`https://gweb-paper-signals.firebaseio.com/signals/your-signalid-here.json`

if your unique id is wrong, then it just returns null, instead of a valid json object so you then get the parseObject() error in the debug log.

Does 

`https://gweb-paper-signals.firebaseio.com/signals/politely-appellate-hippopotamus.json `

return valid json data if you check it in a browser or just say null?

dfashbaugh, 2017-12-05:
@mintisan Does the system spend a long time at the "connecting to gweb-paper-signals.firebaseio.com" or does it seem to immediately fail?

When you're connected to your phone hotspot with another computer are you able to see the JSON data from: https://gweb-paper-signals.firebaseio.com/signals/politely-appellate-hippopotamus.json

I'm not totally familiar with Shadowsocks. But your setup looks good to me so far.

As @tomvdb noted above, there is definitely data at your signal ID URL.

mintisan, 2017-12-05:
@tomvdb Yes, I can open [URL](https://gweb-paper-signals.firebaseio.com/signals/politely-appellate-hippopotamus.json) on my iPhone which acted as proxy. 

And I will check its return value on ESP8266 with UART log.

Thanks.

mintisan, 2017-12-05:
@tomvdb  @dfashbaugh  Maybe I find the reason why  I can't access [URL](https://gweb-paper-signals.firebaseio.com/signals/your-signalid-here.json) for ESP8266.( I can open URL on my iPhone.)

So the proxy configuration need to be modified for hotspot.

I will have a try firstly ...

Thanks so much, @tomvdb  @dfashbaugh

Wanguy, 2017-12-07:
Perhaps It can't connect to `console.firebase.google.com` while using your hotspot. Although you can open the URL on your phone which acted as the proxy when you share your hotspot.

mintisan, 2017-12-09:
After I changed the configuration. It works for me now. :D

```
P address: 172.24.1.23
connecting to gweb-paper-signals.firebaseio.com
requesting URL /signals/politely-appellate-hippopotamus.json
request sent
HTTP/1.1 200 OK

Server: nginx

Date: Sat, 09 Dec 2017 07:39:41 GMT

Content-Type: application/json; charset=utf-8

Content-Length: 503

Connection: close

Access-Control-Allow-Origin: *

Cache-Control: no-cache

Strict-Transport-Security: max-age=31556926; includeSubDomains; preload

headers received
Timer: 30
```

closed 2017-12-09 by mintisan

nicolehe, 2017-12-11:
Glad it's working @mintisan! I'd recommend deleting this signal and creating a new one with a new unique ID for security reasons. :)