googlecreativelab/paper-signals #11: Can't get Google assistant to launch paper signals

yaki4, 2017-12-07:
Hi, i'm not sure that this is the right place to ask this but when i say to my assistant "talk to paper signals", he just respond with an internet link or some pictures of the project. 

Since i'm french, i tried the french version for this sentence but it doesn't work either. 
Am i missing something ?

nicolehe, 2017-12-07:
Hi @yaki4! Are you doing this on a Google Home or Google Assistant on your phone? If it's the Assistant, can you take a screenshot?

yaki4, 2017-12-07:
I've tried it with both Google Home and Google Assistant. 
You can find the screenshots for the assistant below.
[image]
[image]

nicolehe, 2017-12-07:
Unfortunately, because Paper Signals is only supported in English at this time, if your language settings are in French, you won't be able to use the app. If you switch your language settings to English, it should work. Sorry!

yaki4, 2017-12-07:
Oh yeah that make sense. 

Thanks for your response and this great experiment.

closed 2017-12-07 by yaki4