googlecreativelab/paper-signals #15: Unable to compile code

karanrajpal14, 2017-12-14:
I'm unable to compile the code from here. 
The first error is: 
`APICalls.h:23:18: fatal error: Time.h: No such file or directory`

Not a big deal. Apparently, I had to change it to `sys/time.h` and it worked. The error did go away but, a new error was thrown later.
```
APICalls.cpp:275: error: no 'void PaperSignals::StretchBreakExecution(String)' member function declared in class 'PaperSignals'
 void PaperSignals::StretchBreakExecution(String JSONData)
                                                         ^
sketch/APICalls.cpp: In member function 'void PaperSignals::ParseIntentName(String, String)':
APICalls.cpp:692: error: 'StretchBreakExecution' was not declared in this scope
     StretchBreakExecution(JSONData);
                                   ^
Multiple libraries were found for "Servo.h"
 Used: /root/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/Servo
 Not used: /home/karan/arduino-1.8.5/libraries/Servo
exit status 1
no 'void PaperSignals::StretchBreakExecution(String)' member function declared in class 'PaperSignals'
```
I did check the code and StretchBreakExecution was indeed present in APICalls.cpp but it still does not recognize it. I did try to check if something was wrong with the braces and the indentation but, I'm afraid, that's not the case. Any suggestions?

dfashbaugh, 2017-12-14:
Thanks for pointing this out. Looks like one of the files wasn't committed when making the change to include StretchBreak. A pull request is active for this, so the code will be fixed very soon. 

In the meantime, you can pull this branch: https://github.com/dfashbaugh/paper-signals/tree/FixStretchBreak

nicolehe, 2017-12-14:
Hi @karanrajpal14, this should now be fixed, so if you redownload the files or pull from master it should work. Thanks for fixing @dfashbaugh!

karanrajpal14, 2017-12-14:
Thank you so much for fixing it @dfashbaugh 
@nicolehe It's working now :D

closed 2017-12-14 by karanrajpal14