googlecreativelab/paper-signals #17: Countdown time, only counting down seconds

max-epstein, 2017-12-14:
When I say "Ask paper signals to set a timer for 60 seconds," all is well. It counts down. Same goes for "120 seconds" ("one-twenty", not "one hundred twenty"). However if I say "Ask paper signals to set a timer for one minute" it does not work. The servo wobbles back and forth, as it does when you first set a timer, but does not follow through on the countdown. I've tried various amounts of time in both seconds and minutes, and whatever the variation, the problem still persists. Yes to seconds, no to minutes. The issue also seems to exist when asking for longer countdowns, such as an hour, or to a particular time or date. I recognize that for some of these longer countdowns I may not be aware that the servo is moving since I'd imagine it's moving so slowly.

nicolehe, 2017-12-14:
Thanks for finding this, @max-epstein - we need to implement handling for other time units besides seconds. Hope to have a fix today or tomorrow.

Pomezny, 2017-12-15:
@nicolehe I implemented crude conversion for timer in my PR.

nicolehe, 2017-12-18:
HI @max-epstein, this is now fixed with #19, as well as some backend changes.
@Pomezny, we made some changes to the backend (and a small change to the Arduino code in #19) to make sure it works with the voice interaction.

closed 2018-04-02 by dfashbaugh