googlecreativelab/paper-signals #21: Three word identifier from Google Assistant

mactishp, 2018-01-03:
Hi,

I am trying out this project but I get stuck retrieving the three word identifier from Google Assistant as described in the steps. (it doesn't work) I attached a screenshot of this step. Can anybody help me with this?
[image]

nicolehe, 2018-01-03:
Hi @mactishp, can you tell me what it says when you're trying?

mactishp, 2018-01-03:
Hi @nicolehe, 
It simply returns a link to gitHub. (see attached screenshot)
Some sidenotes maybe: I'm trying it on IOS with the Google Allo app (where Google Assistant is integrated in)
[image]

nicolehe, 2018-01-03:
Are you already in the Paper Signals app?

Can you try this:

"Talk to Paper Signals."
and then:
"Create a new signal."

mactishp, 2018-01-03:
I can't follow you there. What do mean with the Paper Signals app? Isn't that just Google Assistant?
In Google Assistant I tried those 2 sentences as is described in the step but Google Assistant just returns a link to a webpage.

nicolehe, 2018-01-03:
Yes - within the Google Assistant, can you try saying "Talk to Paper Signals" and then send a screenshot?

mactishp, 2018-01-03:
Here you go:
[image]

nicolehe, 2018-01-03:
Okay, I think this is the same issue as #11 - is your Assistant set to a language that's not English? Paper Signals is only supported in English, so if your Assistant is set to a different language it won't be able to access it.

mactishp, 2018-01-03:
Sounds logically. But unfortunately, I was already corresponding in English with Google Assistant since Google Assistant doesn't support my country language (Dutch). I even put my phones language to English, reinstalled Google Assistant but it still doesn't work.

nicolehe, 2018-01-03:
Hmm....does it still say "Zeg iets..." when you set the Assistant to English?

mactishp, 2018-01-03:
No, even that is English now.

nicolehe, 2018-01-03:
Do you mind sending another screenshot saying "Talk to Paper Signals" with the app in English?

mactishp, 2018-01-03:
[image]

nicolehe, 2018-01-03:
Thanks, I'll look into it and get back to you.

KOstasHatzis, 2018-01-03:
Try to write the message instead. Let's see what happens...

mactishp, 2018-01-03:
Oh yeah, I got tired a while a go of using the speaking option :D So I was writing the message all this time. I assumed it would/should not make a difference? 
But now I've tried both (speaking/writing) and it gives the same as last screenshot shows. 
It would be nice if you could find what isn't working here since we intend to use this project in a larger marketing workshops towards our customers ;)

KOstasHatzis, 2018-01-03:
Kill the app and install it again. Now that you have the language in English.

KOstasHatzis, 2018-01-03:
I assume from the screenshots that you work on an android phone, right?

mactishp, 2018-01-03:
Yeah, I already reinstalled it (see 4 posts back) but did it again anyways. But no change. 
No, I'm trying it on IOS iPhone.

KOstasHatzis, 2018-01-03:
I have 2 languages in my mob too. Greek and English. And it works just fine.

mactishp, 2018-01-30:
@nicolehe , it still isn"t functioning, any new solutions maybe?

jduitsman, 2018-03-28:
I have the same issue. Only getting links to website's and no unique 3-word identifier. Tried everything.

App is in English, writing, speaking, on phone, on mac etc etc. can't getting it to work. ;-((

nicolehe, 2018-06-12:
Hi @mactishp, @jduitsman, apologies for the long delay. Finally got confirmation that this is not currently supported in the Netherlands. The language when you try to invoke the app should now reflect that. Sorry about that!

closed 2018-06-12 by nicolehe

mactishp, 2018-06-12:
Half a year later is a little bit too late for me...
Btw, it’s Belgium, not The Netherlands.

muthhus, 2019-01-19:
I am getting same issue and unable to move further.
[image]

cipiger, 2019-02-10:
How about the usage in Germany? I do have the same problem, although I changed my language settings to englisch (us).

lromor, 2021-01-18:
I have the same issue. Any news?

cipiger, 2021-01-20:
Sorry, nio. This google service was not supported in germany.
 

Gesendet: Montag, 18. Januar 2021 um 16:34 Uhr

michaelshorter, 2021-07-22:
Hi all,

I'm also having the exact same problem. Running Google Assistant in English and in the UK. Has anyone got to the bottom of this yet?