googlecreativelab/paper-signals #26: Upload error on Nodemcu ESP 8266

sarrocks1, 2018-04-01:
I tried uploading the code with the correct signal id at all different baud rates.
I am getting these errors all the time -

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 921600"

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\admin\AppData\Local\Temp\arduino_cache_418121\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_sdk,UploadSpeed_921600_5f03886daecb6caa97d15be6e5b3cbfc.a
Sketch uses 348960 bytes (33%) of program storage space. Maximum is 1044464 bytes.
Global variables use 38264 bytes (46%) of dynamic memory, leaving 43656 bytes for local variables. Maximum is 81920 bytes.
**warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed**

Have I made any mistakes?

dfashbaugh, 2018-04-02:
Seems like its getting past compile and failing to upload.

Are you sure you have the silicon labs driver installed? 
Here is the software if you need it: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

Looks like you are in Windows, so you should see a COM port show up once the driver is installed and the NodeMCU board is plugged in. Are you selecting the correct COM port for upload in Arduino? 

You should be able to see the COM port for your board under Tools -> Port.

dfashbaugh, 2018-04-02:
Also, is it failing to upload all of the time or just some of the time?
Sometimes lowering the Upload Speed can help depending on your USB Cable / ESP Circuit. 

Are you sure that your USB Cable has data lines and is not just for charging?

sarrocks1, 2018-04-02:
@dfashbaugh  Thank You! Apparently I had connected the signal pin at the wrong place
For anyone who uses NodeMCU ESP 8266 , use D5 for the servo signal pin. And select either NodemMCU 1.0 or Adafruit feather Huzzah board in the arduino ide.
Thanks!!!

closed 2018-04-02 by sarrocks1