googlecreativelab/paper-signals #27: DynamicJsonBuffer Issue

levimail12, 2018-06-08:
I followed all the directions when on Connect your Paper Signal when I got to step 7 and I tried to upload I got an error proclaiming

> 'DynamicJsonBuffer' was not declared in this scope

wonder what I did wrong, a or more likely what information might be usefull in solving this

dfashbaugh, 2018-06-09:
Hey @levimail12 , Looks like ArduinoJSON released a new version that is not compatible with Paper Signals at the moment. To get compiling, you'll need to rollback a version.

To do this, Open up Arduino. Then go to Sketch->Include Library->Manage Libraries...

Then search for ArduinoJSON.

When you find the library, just use the drop down in the bottom left to select a slightly older version and click "Update."

[image]

mattboh9, 2020-04-18:
Is this still a problem because I’m having a similar situation that when I get to step 7 I get a compile error.

closed 2023-09-01 by mrayinteractive