googlecreativelab/paper-signals #31: Connecting to gweb-paper-signals.firebaseio.com Connection failed

ShaylinChetty, 2019-01-04:
Hi,

Similar to #8 upon uploading the code I get the following error in the console

`connecting to gweb-paper-signals.firebaseio.com
connection failed
parseObject() failed
Default`

The wifi connection is made successfully and all proxy/port blocking setting have been disabled. Accessing https://gweb-paper-signals.firebaseio.com/signals/my-signal-id.json works fine leading me to believe this is not an issue with my ID.

Thanks

archetana, 2019-01-09:
The issue got resolved, I had installed the ESP8266 board version 2.5.0, once I removed and installed 2.3.0, papersignals is working fine now.

ShaylinChetty, 2019-01-10:
Rolling back to 2.3.0 has resolved the issue for me as well. Thanks for the help.

closed 2019-01-10 by ShaylinChetty