googlecreativelab/paper-signals #33: Does this still work?

amweiss, 2019-09-07:
The first time I did "Talk to paper signals" there were already 12 signals and it seems like none of them work and I cannot get any of them to even run the servo test. I've tried about 6 of the setup codes, but no difference. Is the service still running?

amweiss, 2019-09-07:
Nevermind, the board package was the wrong version as well.
For anyone else still trying, be sure to pick the board package and arduinojson package shown in the screenshots to have it actually work.

closed 2019-09-07 by amweiss

BritneyMuller, 2020-01-28:
@amweiss Hey Adam,

Also, experiencing problems connecting any of the 12 pre signals as well despite using the board package in the screenshots. Did you end up using one of the original 12 pre-installed signal ID's or did you have success creating a new one?

Can't get either to work at the moment and any insights would be greatly appreciated 🙏

amweiss, 2020-01-28:
Hey @BritneyMuller I got past the board package compiling and being able to send commands to the board, but none of the actual voice commands or signals seem to work anymore :(

So I can use the existing python to write a program and push to the board, but it has to be using new endpoints, storage, voice commands, etc instead of anything that is described on the site.

BritneyMuller, 2020-01-29:
Well...shit. 

Really really appreciate you letting me know @amweiss! Thank you! I'll try to get a PR submitted in the readme to let others know.

BritneyMuller, 2022-10-26:
Still really sad this doesn't work anymore. 🙃