googlecreativelab/paper-signals #38: Signals not communicating

Andrewpk, 2020-02-21:
Signals firebase json is showing that our device has been tested as usable and reports as the proper type/name, but the device is not communicating.

### Process to reproduce:
- Create a paper signal
- Get the 3 word signal ID
- upload the code to your esp8266 with three word signal ID
- Use google assistant to test your paper signal once the code has been uploaded to your esp, your device has rebooted, and your device has properly joined the wifi network.

### Expected Result:
- Servo motor sweeps

### Actual Result:
- Nothing happens

savannahostrowski, 2020-05-09:
I'm also experiencing this so if anyone can provide insight, that'd be super helpful! 

Is this repo being maintained?

Andrewpk, 2020-05-09:
It does not appear this is being maintained at all.
The google assistant service itself seems very broken.

mattboh9, 2020-05-12:
I got my paper signal to work with very similar problems that other people were facing. If you have any questions feel free to ask me, I’d love to help.

mattboh9, 2020-05-12:
Have you guys changed your downloads to the exact versions that are on the paper signals website?

nikitasarker, 2020-08-08:
Also having this problem using the exact versions from the website

mattboh9, 2020-08-08:
It’s been a while. But for me I had to download the exact versions that the
paper signal walk through had. Then I had to update the java on my laptop.
Let me look further into it to refresh my memory and I’ll get back to you.

avroncohen, 2020-09-09:
I was having a similar problem. Changed the esp8266 board manager to version 2.3.0 (used in the instructions) and it worked.

The google assistant app is showing a high number of signals. Thought it would only show signals related to my google account.

The signal ID was four words and not three stated in the instructions.

Hope this helps.

closed 2023-09-01 by mrayinteractive